gazebosim/sdformat #814: preserveFixedJoint flag in sdf::readFile

varunagrawal, 2022-01-04:
## Desired behavior
Would having a `preserveFixedJoint` flag in the `sdf::readFile` function be a good idea? Our projects would like to preserve fixed joints but we would like to avoid having to manually edit each and every SDF/URDF file that our users use.

Since we parse the SDF/URDF file directly using the `readFile` function, it makes sense to have a flag there with a sensible, backwards-compatible default.

## Alternatives considered
The alternative would be to manually update each SDF/URDF file to add the necessary `<gazebo>` tags. This is cumbersome and error-prone in case someone forgets to do it.

## Implementation suggestion
I can open a PR for this and we can discuss the overall design there. I don't believe it will be a big change, just exposing some functionality programmatically.

## Additional context

scpeters, 2022-01-04:
I think you may want to add it to [ParserConfig.hh](https://github.com/ignitionrobotics/sdformat/blob/sdf12/include/sdf/ParserConfig.hh)

varunagrawal, 2022-01-04:
Great! Will there be a plan to backport this to SDFormat10 and SDFormat11?

scpeters, 2022-01-04:
> Great! Will there be a plan to backport this to SDFormat10 and SDFormat11?

it is already in libsdformat11, and libsdformat10 has reached end-of-life

* https://github.com/ignitionrobotics/sdformat/blob/sdf11/include/sdf/ParserConfig.hh

azeey, 2022-01-14:
Closed by #815

PR #815 (2022-01-14): Add ParserConfig flag for preserveFixedJoint
# 🎉 New feature

Closes #814

## Summary

Added a new boolean flag `preserveFixedJoint` to `ParserConfig` and `updated parser_urdf.cc` to use this flag.

## Test it

The test `CheckParserConfig_preserveFixedJoint` in `parser_urdf_TEST.cc` demonstrates the functionality. You essentially set the flag in a `ParserConfig` object and pass it to the appropriate URDF parsing function.

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests
- [x] Added example and/or tutorial
- [x] Updated documentation (as needed)
- [x] Updated migration guide (as [... cut]
closed 2022-01-14 by azeey