gazebosim/sdformat #881: ParserConfig is sometimes ignored

jwnimmer-tri, 2022-03-15:
## Environment
* OS Version: Ubuntu 20.04
* Source build, using tag `sdformat12_12.0.0`.
    
## Description
* Expected behavior: When I call `sdf::Root::Load(my_filename, my_parser_config)`, I expect the `ParserConfig` that I passed in to be used for the entire parse.
* Actual behavior: Sometimes, the `ParserConfig::GlobalConfig()` is used instead, especially for included files.

## Steps to reproduce

Refer to the PR at https://github.com/RobotLocomotion/drake/pull/16791.  In that PR, I replace the `GlobalConfig` accessor function with a runtime exception.  Then, a `bazel test //multibody/parsing/...` shows that exception tripping, even though I've always passed a `ParserConfig` to every function that I've called.

The PR has a sdformat patch for the instances that I was able to find.  Check the file `drake/tools/workspace/sdformat/patches/fix_broken_config.patch` from that PR.  I'd appreciate if you could incorporate that patch upstream here.

There may be more instances lurking, but those were only the ones I found.

One way you could audit for more bugs beyond what I found is to momentarily mark the dangerous functions (the ones which locally default to the the GlobalConfig) as deprecated, and then run a build and inspect all of the deprecation warnings at the call sites.  Any call sites that have a `_config` argument in scope but don't pass it along are suspicious.

jwnimmer-tri, 2022-03-16:
In related news, I think it's a mistake that the `CustomModelParser` signature does not pass the in-effect `ParserConfig` along to the user's custom parser.  As the user parses their nested model, they will also need to find files, look up URIs, check the warnings policy, etc.  The user can't just use the GlobalConfig because a different config might have been chosen for this parse.

scpeters, 2022-03-16:
one of the fixes from your patch has already been applied in #824; I'm taking a look at the other portions

scpeters, 2022-03-16:
I've applied your patch and some additional improvements in https://github.com/ignitionrobotics/sdformat/pull/883, though there is still some additional work to do. For testing, I made a test branch using your suggestion of deprecating `parser.hh` functions that don't use `ParserConfig` (https://github.com/ignitionrobotics/sdformat/commit/34383d5c655eb54930d23c2780828d20b9f4f459).

scpeters, 2022-03-18:
another follow-up in #885

PR #883 (2022-03-16): Use ParserConfig more in parser.cc
# 🦟 Bug fix

Part of https://github.com/ignitionrobotics/sdformat/issues/881

## Summary

As noted in #881, `ParserConfig` objects are not always propagated even when they are available. A recent partial fix for this was recently merged in #824, and a patch by @jwnimmer-tri was suggested in https://github.com/RobotLocomotion/drake/pull/16791, which has been applied to this branch in https://github.com/ignitionrobotics/sdformat/commit/58198fa75b1531fb03827d92d11c92e5c1b2c3db. Additionally, several private helper functions in parser.cc are updated in https://gi [... cut]