gazebosim/sdformat #935: Link needs helper function to get Inertial relative to a named frame

scpeters, 2022-04-04:
## Desired behavior

Currently the `Link::Inertial()` accessor returns an `Inertial` object with the `Pose` specified relative to the link frame. Since there is no `sdf::Inertial` class with a `SemanticPose()` API, it would be nice to provide a `Link::ResolveInertial(ignition::math::Inertiald &_inertial, const std::string &_resolveTo = "")` method that would resolve the `Inertial`'s `Pose` to a specified frame, such as `__model__`.

## Alternatives considered

We could create an `sdf::Inertial` object or just leave the API as is and provide documentation for how to resolve multiple `Inertial` objects to a common frame.

## Implementation suggestion

~~~
Errors Link::ResolveInertial(
  ignition::math::Inertiald &_inertial,
  const std::string &_resolveTo = "")
{
  ignition::math::Pose3d linkPose;
  auto errors = this->SemanticPose().Resolve(linkPose, _resolveTo);
  if (errors.empty())
  {
    _inertial.SetPose(linkPose * _inertial.Pose());
  }
  return errors;
}
~~~

## Additional context

azeey, 2022-04-25:
This sounds useful and the implementation suggestion looks good. There's precedent for doing this type of thing in `JointAxis::ResolveXyz`. This would also mean we can add `//inertal/pose/@relative_to` easily.

scpeters, 2022-05-11:
see #1012

PR #1012 (2022-05-23): Add Link::ResolveInertial API
# 🎉 New feature

Closes #935

## Summary

The `Link::Inertial()` method returns an Inertial object with a Pose
specified relative to the link frame. This adds a new method for
resolving the Inertial's Pose to a specified frame. The method
is demonstrated in the `ign sdf --inertial-states` command code.

## Test it

Run the `UNIT_ign_TEST` and confirm that `ign sdf --inertial-stats` test still passes.

## Checklist
- [X] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration g [... cut]
commit a428e2ad (2022-05-23): Add Link::ResolveInertial API (#1012)

The Link::Inertial() method returns an Inertial object with a Pose
specified relative to the link frame. This adds a new method for
resolving the Inertial's Pose to a specified frame. The method
is demonstrated in the `ign sdf --inertial-states` command code.

Fixed #935.
closed 2022-06-06 by azeey