gazebosim/sdformat #977: `Model::Load` does not set the `poseGraph` graph

varunagrawal, 2022-04-13:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build? Source build from branch `sdf12`, commit `6404afdf`
    
<details>
  
```
# paste log here
```

</details>

## Description
Apologies if this is not a code bug, but I couldn't find anything in the docs and looking at the source code makes me think this is a valid bug.

* Expected behavior: Calling `Model::Load` should set the `poseGraph` graph so that calling `link->SemanticPose().Resolve(pose)` works as expected.
* Actual behavior: We get an error `SemanticPose has invalid pointer to PoseRelativeToGraph.` and so the pose resolution is invalid.

## Steps to reproduce

I wrote up a unit test in `parser_TEST.cc` to reproduce this:

```cpp
#include "sdf/Model.hh"
#include "sdf/World.hh"
#include "sdf/Link.hh"
TEST(Parser, ResolvePose)
{
  const std::string testString = R"(<?xml version="1.0" ?>
    <sdf version="1.9">
      <world name="default">
        <frame name="world"/>
        <model name="simple">
          <link name="link_0">
            <pose relative_to="world">0 0 0 0 0 0</pose>
          </link>
        </model>
      </world>
    </sdf>)";

  sdf::Errors errors;
  sdf::SDFPtr sdf = InitSDF();
  sdf::ParserConfig config;
  sdf::readString(testString, config, sdf, errors);

  sdf::Model model;
  model.Load(sdf->Root()->GetElement("model"));
  EXPECT_EQ("__default__", model.Name());

  sdf::World world;
  world.Load(sdf->Root()->GetElement("world"));
  model = *world.ModelByIndex(0);
  EXPECT_EQ("simple", model.Name());

  auto link = model.LinkByName("link_0");

  auto raw_pose = link->RawPose();
  auto semanticPose = link->SemanticPose();
  errors = semanticPose.Resolve(raw_pose);

  EXPECT_EQ(0, errors.size());
  if (errors.size() > 0) {
    for(size_t i=0; i<errors.size(); i++) {
      std::cout << errors[i].Message() << std::endl;
    }
  }
}
```

The above test should pass. Maybe there is something wrong in the way I am defining my SDF file? Either way it would be great to know how to fix this and this could serve to help someone else in the future.

varunagrawal, 2022-04-14:
I also tried using the SDF example in the [tutorial](http://sdformat.org/tutorials?tut=pose_frame_semantics&ver=1.7&cat=specification&) and I still get the issue:

```cpp
#include "sdf/Model.hh"
#include "sdf/World.hh"
#include "sdf/Link.hh"
TEST(Parser, ResolvePose)
{
  const std::string testString = R"(<?xml version="1.0" ?>
    <sdf version="1.7">
  <model name="pendulum_with_base">
    <link name="base">
      <pose>0 0 0.3   0 0 0</pose>
    </link>
    <link name="pendulum">
      <pose relative_to="joint">
        0 0 -0.5 0 0 0
      </pose>
    </link>
    <joint name="joint" type="revolute">
      <parent>base</parent>
      <child>pendulum</child>
      <pose relative_to="base">
        0 0 0.73 1.57 0 0
      </pose>
      <axis>
        <xyz>1 0 0</xyz>
      </axis>
    </joint>
  </model>
</sdf>)";

  sdf::Errors errors;
  sdf::SDFPtr sdf = InitSDF();
  sdf::ParserConfig config;
  sdf::readString(testString, config, sdf, errors);

  sdf::Model model;
  model.Load(sdf->Root()->GetElement("model"));

  auto link = model.LinkByName("base");

  auto raw_pose = link->RawPose();
  auto semanticPose = link->SemanticPose();
  errors = semanticPose.Resolve(raw_pose);

  EXPECT_TRUE(0 == errors.size());
  if (errors.size() > 0) {
    for(size_t i=0; i<errors.size(); i++) {
      std::cout << errors[i].Message() << std::endl;
    }
  }
}

azeey, 2022-04-14:
The main entry point for using the DOM classes is `sdf::Root` and as of `libsdformat` 11.0, this is where the `poseGraph` is set. In your case, the code would look like
```c++
...
sdf::Errors errors;
sdf::ParserConfig config;
sdf::Root root;
root.LoadSdfString(testString, config);
const sdf::Model * model = root.Model();
...
```

varunagrawal, 2022-04-14:
You are a gem, @azeey! Worked perfectly.

closed 2022-04-14 by varunagrawal