gazebosim/sdformat #1007: URDF to SDF conversion ignores links with small mass

traversaro, 2022-05-02:
Spin-off of https://github.com/ignitionrobotics/sdformat/issues/199 for visibility. Quoting from https://github.com/ignitionrobotics/sdformat/issues/199#issuecomment-612375956 : 

> **Original comment by Kaelin (Bitbucket: [Kaelin](https://bitbucket.org/%7B8d4c9653-2af7-4ce2-bb67-6cb8750d5868%7D/)).**
> 
> I happen to have run into a variant of this issue today and spent a while debugging [it.](http://it.In) In addition to the case of having not declared an `inertia` node in the XML at all, this also occurs if you have a `mass` value which is too small. A mass value of `0.00001` (`1e-5`) works fine, but if you decrease it to `0.000001` (`1e-6`), the link is silently omitted. This is pretty dangerous, since it is easy to gloss over an additional `0` when scanning for issues. I saw that having _no mass_ was disallowed, so my natural inclination was to make it _some very small number_, and happened to hit too many zeroes.
> 
> An error in this case, even if massless links aren’t implemented, would be much appreciated.

traversaro, 2022-05-02:
The issue is due to the following line:
* https://github.com/ignitionrobotics/sdformat/blob/sdformat12_12.4.0/src/parser_urdf.cc#L2644

it may not be immediatly clear why ``, but this come from the fact that the second argument of `ignition::math::equal` is not specified and by default it is 1e-6:
* https://github.com/ignitionrobotics/ign-math/blob/ignition-math6_6.10.0/include/ignition/math/Helpers.hh#L557

azeey, 2023-03-23:
Fixed by https://github.com/gazebosim/sdformat/pull/1238

closed 2023-03-23 by azeey

aaronchongth, 2023-03-24:
> Fixed by #1238

More specifically, https://github.com/gazebosim/sdformat/pull/1238 does not allow link mass with value of `<= 0`, without any epsilon values, https://github.com/gazebosim/sdformat/pull/1238/files#diff-2bc5ca23bcfc66fe173f513399bec8065b1cb4e607d741566517d7368e6bce0fR2670