gazebosim/sdformat #1046: GUI DOM object doesn't include user camera configuration

scpeters, 2022-06-09:
## Description

The GUI DOM object defined in [sdf/Gui.hh](https://github.com/gazebosim/sdformat/blob/sdformat12_12.5.0/include/sdf/Gui.hh) doesn't include the parameters related to the user camera in [//gui/camera](https://github.com/gazebosim/sdformat/blob/sdformat12_12.5.0/sdf/1.9/gui.sdf#L9-L59).

chapulina, 2022-06-14:
We should consider deprecating everything inside `//gui` except for `//gui/plugin`. With the move to [Gazebo GUI](https://github.com/gazebosim/gz-gui/), most of the GUI configurations are plugin-specific and end up inside plugin tags.

There are ways in Gazebo to propagate these parameters from `//gui` to the plugins through the ECM, but it's quicker to iterate and add elements if they're not part of the SDF spec. It's also more organized, because the configuration is closer to the plugins that will use it. Finally, the current `//gui` element makes assumptions that limit its use (i.e. there's a single user camera / view).

I don't know whether users of SDF besides Gazebo are using this element.

scpeters, 2022-06-14:
ok, that makes sense

closed 2022-06-14 by scpeters