gazebosim/sdformat #1124: Linking problem due to tinyxml2

julianoes, 2022-08-31:
## Environment
* OS Version: Debian 11 Bullseye
* Source build using colcon as documented in https://gazebosim.org/docs/fortress/install_ubuntu_src

## Description

I'm trying to build Ignition Gazebo 6 on Debian. The colcon build fails with a linking error for sdformat12. It looks like something goes wrong with the linking of tinyxml2 symbols. Presumably the tinyxml2 library installed on my system conflicts with tinyxml2 in tree [in gz-common](https://github.com/gazebosim/gz-common/tree/ign-common4/graphics/src/tinyxml2).

I tried to build everything without having tinyxml2 installed system-wide but that does not work either as it is required by sdformat12.

## Steps to reproduce

```
vcs import < collection-fortress.yaml
colcon graph
colcon build --cmake-args -DBUILD_TESTING=OFF --merge-install
```

Should I provide a debian based Dockerfile to reproduce it?

## Output

Colcon build without `libtinyxml2-dev` installed system-wide:
```
Starting >>> sdformat12                                                              
--- stderr: sdformat12                                                 
CMake Error at /home/julianoes/src/other/ign-workspace/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:59 (message):
  -- BUILD ERRORS: These must be resolved before compiling.
Call Stack (most recent call first):
  CMakeLists.txt:115 (ign_configure_build)

CMake Error at /home/julianoes/src/other/ign-workspace/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:61 (message):
  -- 	Missing dependency [TINYXML2]
Call Stack (most recent call first):
  CMakeLists.txt:115 (ign_configure_build)
```

Build with `tinyxml2` installed:

```
/usr/bin/ld: CMakeFiles/sdformat12.dir/urdf/urdf_parser/model.cpp.o: in function `tinyxml2::XMLDocument::ClearError()':
/usr/include/tinyxml2.h:1871: undefined reference to `tinyxml2::XMLDocument::SetError(tinyxml2::XMLError, int, char const*, ...)'
/usr/bin/ld: ../lib/libsdformat12.so.12.5.0: hidden symbol `_ZN8tinyxml211XMLDocument8SetErrorENS_8XMLErrorEiPKcz' isn't defined
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
gmake[2]: *** [src/CMakeFiles/sdformat12.dir/build.make:1247: lib/libsdformat12.so.12.5.0] Error 1
```

I've initially asked in https://github.com/gazebosim/gz-common/issues/430 but I'm adding the issue here as well, as it's sdformat failing to link.

scpeters, 2022-08-31:
> Should I provide a debian based Dockerfile to reproduce it?

yes please, that would be helpful.

What is the version of tinyxml2 installed in your system?

julianoes, 2022-09-01:
The version should be: https://packages.debian.org/bullseye/libtinyxml2-8

julianoes, 2022-09-19:
Thanks for the hint @scpeters. With a proper directory, I was able to reproduce it:

```
FROM debian:bullseye

RUN apt-get update \
    && apt dist-upgrade -y \
    && apt install -y \
        git \
        wget \
        python3-pip \
        python-is-python3 \
        python3-distutils \
        build-essential \
        cmake \
        libtinyxml2-dev \
        libavutil-dev \
        libavcodec-dev \
        libavformat-dev \
        libavdevice-dev \
        libfreeimage-dev \
        libeigen3-dev \
        libgts-dev \
        libprotobuf-dev \
        libprotoc-dev \
        protobuf-compiler \
        protobuf-c-compiler \
        pkg-config \
        ruby

RUN python -m pip install vcstool
RUN python -m pip install colcon-common-extensions

RUN mkdir -p /ws/src
WORKDIR /ws/src
RUN wget https://raw.githubusercontent.com/ignition-tooling/gazebodistro/master/collection-fortress.yaml
RUN vcs import < collection-fortress.yaml
RUN colcon graph
RUN colcon build --cmake-args -DBUILD_TESTING=OFF --merge-install
```

The output is then:

```
...
Step 10/10 : RUN colcon build --cmake-args -DBUILD_TESTING=OFF --merge-install
 ---> Running in 5daaba0e273e
Starting >>> ignition-cmake2
Starting >>> ignition-tools
Finished <<< ignition-cmake2 [0.41s]
Starting >>> ignition-math6
Starting >>> ignition-utils1
Finished <<< ignition-tools [1.01s]
Finished <<< ignition-utils1 [1.65s]
--- stderr: ignition-math6
CMake Warning at /ws/src/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:55 (message):
   CONFIGURATION WARNINGS:
   -- Swig is missing: Language interfaces are disabled.
   -- pybind11 is missing: Python interfaces are disabled.
Call Stack (most recent call first):
  CMakeLists.txt:104 (ign_configure_build)

---
Finished <<< ignition-math6 [7.31s]
Starting >>> ignition-common4
Starting >>> ignition-msgs8
Starting >>> sdformat12
[Processing: ignition-common4, ignition-msgs8, sdformat12]
Finished <<< ignition-common4 [51.8s]
Starting >>> ignition-plugin1
Finished <<< ignition-plugin1 [7.29s]
Starting >>> ignition-rendering6
--- stderr: sdformat12
CMake Warning at /ws/src/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:55 (message):
   CONFIGURATION WARNINGS:
   -- Python psutil package not found. Memory leak tests will be skipped
Call Stack (most recent call first):
  CMakeLists.txt:115 (ign_configure_build)

/ws/src/sdformat/src/urdf/urdf_parser/world.cpp:36: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
   36 | #pragma warning(push, 0)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/world.cpp:72: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
   72 | #pragma warning(pop)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/urdf_sensor.cpp:36: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
   36 | #pragma warning(push, 0)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/urdf_sensor.cpp:395: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
  395 | #pragma warning(pop)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/urdf_model_state.cpp:36: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
   36 | #pragma warning(push, 0)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/urdf_model_state.cpp:166: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
  166 | #pragma warning(pop)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/twist.cpp:36: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
   36 | #pragma warning(push, 0)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/twist.cpp:81: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
   81 | #pragma warning(pop)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/pose.cpp:36: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
   36 | #pragma warning(push, 0)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/pose.cpp:133: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
  133 | #pragma warning(pop)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/joint.cpp:36: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
   36 | #pragma warning(push, 0)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/joint.cpp:694: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
  694 | #pragma warning(pop)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/link.cpp:36: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
   36 | #pragma warning(push, 0)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/link.cpp:679: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
  679 | #pragma warning(pop)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/model.cpp:36: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
   36 | #pragma warning(push, 0)
      | 
/ws/src/sdformat/src/urdf/urdf_parser/model.cpp:264: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
  264 | #pragma warning(pop)
      | 
/usr/bin/ld: CMakeFiles/sdformat12.dir/urdf/urdf_parser/model.cpp.o: in function `tinyxml2::XMLDocument::ClearError()':
/usr/include/tinyxml2.h:1871: undefined reference to `tinyxml2::XMLDocument::SetError(tinyxml2::XMLError, int, char const*, ...)'
/usr/bin/ld: ../lib/libsdformat12.so.12.6.0~pre1: hidden symbol `_ZN8tinyxml211XMLDocument8SetErrorENS_8XMLErrorEiPKcz' isn't defined
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
gmake[2]: *** [src/CMakeFiles/sdformat12.dir/build.make:1247: lib/libsdformat12.so.12.6.0~pre1] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:451: src/CMakeFiles/sdformat12.dir/all] Error 2
gmake: *** [Makefile:182: all] Error 2
---
Failed   <<< sdformat12 [1min 4s, exited with code 2]
Aborted  <<< ignition-rendering6 [19.1s]
Aborted  <<< ignition-msgs8 [1min 21s]

Summary: 6 packages finished [1min 29s]
  1 package failed: sdformat12
  2 packages aborted: ignition-msgs8 ignition-rendering6
  3 packages had stderr output: ignition-math6 ignition-rendering6 sdformat12
  7 packages not processed
The command '/bin/sh -c colcon build --cmake-args -DBUILD_TESTING=OFF --merge-install' returned a non-zero code: 2
```

scpeters, 2022-09-21:
I see a clue in the pragma warnings:

~~~
/ws/src/sdformat/src/urdf/urdf_parser/world.cpp:36: warning: ignoring ‘#pragma warning ’ [-Wunknown-pragmas]
   36 | #pragma warning(push, 0)
      | 
~~~

libsdformat has a vendor version of urdfdom and urdfdom_headers in the `src/urdf` folder, which is where this compiler warning is coming from. If `liburdfdom-dev` is installed, then I think it would not have this problem.

This is still a problem, but can you try installing `liburdfdom-dev`, wiping the build folder and trying again?

julianoes, 2022-09-23:
Thanks @scpeters. I think installing `liburdfdom-dev` seemed to fix that part of the build indeed. I assume there should be some configure step that makes sure the system one is available, so that this problem doesn't occur?

The build now fails due to a similar problem than https://github.com/gazebosim/gz-transport/pull/237 but I'll have to create a separate issue for that.

julianoes, 2022-09-23:
Re gz-transport build, follow up in https://github.com/gazebosim/gz-transport/issues/362 and https://github.com/gazebosim/gz-transport/pull/363.

scpeters, 2022-09-23:
> Thanks @scpeters. I think installing `liburdfdom-dev` seemed to fix that part of the build indeed. I assume there should be some configure step that makes sure the system one is available, so that this problem doesn't occur?

you can set the cmake variable `-DUSE_INTERNAL_URDF=OFF` to force using the system installation of urdfdom. Otherwise it uses the vendor version if the external version is not found

julianoes, 2022-09-23:
Ok, that's good to know but I just have the feeling this can really confuse one (like me) thoroughly and there has to be a better way to prevent that sort of trouble, no?

azeey, 2022-11-16:
What's the state of this issue? Is there an action item?

scpeters, 2022-11-16:
> What's the state of this issue? Is there an action item?

if anything, I think it is to improve the documentation and console messaging about the `-DUSE_INTERNAL_URDF=OFF` variable and whether the system version of urdfdom is being used or the from-source fork