gazebosim/sdformat #1125: Clarify definition of //joint/thread_pitch for screw joints

scpeters, 2022-09-01:
## Desired behavior

The units and meaning of the [//joint/thread_pitch](https://github.com/gazebosim/sdformat/blob/sdformat12_12.5.0/sdf/1.9/joint.sdf#L39-L41) element is not defined in the SDFormat specification.

Per [wikipedia](https://en.wikipedia.org/wiki/Screw_thread#Lead,_pitch,_and_starts), the *pitch* or a screw is the axial distance between adjacent threads, while the *lead* of a screw is the axial distance traveled during one complete turn. For single-start threads with only one ridge wrapped around the screw, the *pitch* and *lead* are identical. The *lead* and *pitch* differ however for double-start and other non-single start threads.

From a kinematics perspective, the *lead* is the more precise name for the important parameter than pitch, but for now, let's just document the interpretation of `//joint/thread_pitch` assuming a single-start thread.

## Alternatives considered

## Implementation suggestion

## Additional context

scpeters, 2022-09-01:
For reference, [dartsim](https://github.com/dartsim/dart/blob/main/dart/dynamics/detail/ScrewJointAspect.hpp#L55-L56) defines labels the screw joint pitch as "translational pitch" and from [its test](https://github.com/dartsim/dart/blob/95f065b6879dcdb6522fd543871e90aaed50738c/unittests/integration/test_ScrewJoint.cpp#L62-L64) appears to have units of `meters / revolution`.

Likewise the [screw_pitch](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_screw_joint.html#a7c551ba379c10c64505157d498c83327) parameter in drake is defined as "the amount of translation in meters occuring over a one full revolution".

By contrast, [gazebo classic](https://github.com/osrf/gazebo/blob/gazebo11_11.11.0/gazebo/physics/ScrewJoint.hh#L60-L76) defines thread pitch as "angular motion per linear motion" with units of "rad / m" and some models exist using the `//joint/thread_pitch` with this interpretation.

EricCousineau-TRI, 2022-09-08:
Per http://www.cds.caltech.edu/~murray/mlswiki/images/0/02/Mls94-complete.pdf, p.45 (pdf page 63), "Screws: a geometric description of twists":
> [...] we define the pitch of the screw to be the ratio of translation to rotation, h := d/θ (assuming θ != 0). Thus, the net translational motion after rotating by θ radians is hθ

Not a fan :grimacing: But maybe still good to gather a few more references?

EricCousineau-TRI, 2022-09-08:
Is there any record of the reference that motivated usage of current `//joint/thread_pitch` units? (BitBucket or GitHub PR?, git commit explanation ,etc?)

EricCousineau-TRI, 2022-09-22:
Req. from @azeey is to see what Bullet does.

scpeters, 2022-09-23:
> Req. from @azeey is to see what Bullet does.

Bullet doesn't have a native screw constraint. gazebo-classic uses a 2-dof `btSliderConstraint` (allows translation and rotation about the same axis) and applies the screw constraint using [custom code](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/physics/bullet/BulletScrewJoint.cc#L862-L894)

scpeters, 2022-09-23:
> Is there any record of the reference that motivated usage of current `//joint/thread_pitch` units? (BitBucket or GitHub PR?, git commit explanation ,etc?)

the earliest thing I can find is from 2011 when the Willow Garage fork of Open Dynamics Engine was imported into the gazebo fork in https://github.com/osrf/gazebo/commit/867c7e12317ec15b5c61db67bd14760e943b423c, which included the added `joints/screw.h` header file, which [has not been upstreamed](https://bitbucket.org/odedevs/ode/src/master/ode/src/joints/screw.h) to [odedevs/ode](https://bitbucket.org/odedevs/ode/). That import header file [defines the thread pitch as radians / meter](https://github.com/osrf/gazebo/blame/gazebo11/deps/opende/src/joints/screw.h#L41).

I also just noticed some comments elsewhere in gazebo suggesting that the parameter definition should be inverted to "translation / angle"

* https://github.com/osrf/gazebo/blob/gazebo11/gazebo/physics/ode/ODEScrewJoint.cc#L206-L219

scpeters, 2022-09-23:
I personally prefer units of `meters per revolution`

scpeters, 2022-09-23:
I've opened a pull request to document the units and sign convention of the `//joint/thread_pitch` parameter based on its use in gazebo-classic. I also added a `//joint/screw_thread_pitch` parameter with more sane units and sign convention and deprecated `//joint/thread_pitch`. https://github.com/gazebosim/sdformat/pull/1168

If possible I'd like to merge this soon after the Garden release along with #1117

closed 2022-10-18 by scpeters