gazebosim/sdformat #1129: Pose3 return type not valid for pybind11

willstott101, 2022-09-07:
## Environment
* OS Version: Custom poky-based distro (via Ubuntu 20.04)
* Source or binary build: Source from `1bf1aa4e511d3efe0f97a6c14048a35dbcde26ce` on `sdf13`

## Description
* Expected behavior: Python bindings allow me to get resolved (joint) poses
* Actual behavior: PyBind11 cannot convert the type

## Steps to reproduce
1. Load an sdf using the python module
2. Try to access a method that returns `gz::math::v7::Pose3<double>` such as `SemanticPose.resolve`

## Output
```
>>> p.raw_pose()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: Unable to convert function return value to a Python type! The signature was
	(self: sdformat.SemanticPose) -> gz::math::v7::Pose3<double>
```

## Possible Solution
I suspect gz-math needs to provide python object access as described here: https://github.com/pybind/pybind11/issues/616

I'm going to see if I can get that to work and will post an example if it does - but I don't think I'll have time to put together a proper PR.

ahcorde, 2022-09-07:
Did you try to import `from gz.math import Pose3d` ?

willstott101, 2022-09-07:
Ah no I didn't... that works. Thanks!

Should the `sdformat` library import that automatically?

ahcorde, 2022-09-07:
it seems that is not bringing the dependency, I will try to review this when I have some time.

Feel free to open a PR and tag me if you find a solution.

willstott101, 2022-09-07:
ok failing test here: https://github.com/willstott101/sdformat/runs/8225866161?check_suite_focus=true
Based on: https://github.com/willstott101/sdformat/commit/6e66c67a5af8c435d28e5b9fdadc01acbf5e2cab

This seems to fix it: https://github.com/gazebosim/sdformat/compare/sdf13...willstott101:sdformat:devtool?expand=1
passing test: https://github.com/willstott101/sdformat/runs/8226155319?check_suite_focus=true

ahcorde, 2022-09-07:
Hi @willstott101, 

do you mind to open a PR with your fix ?

willstott101, 2022-09-07:
https://github.com/gazebosim/sdformat/pull/1130 Opened one - but haven't gone through all the red tape yet - might get round to that some time but really not that interested in filling in a bunch of forms

ahcorde, 2022-10-03:
The fix is merged. Closing this

commit 8bde4a9d (2022-09-30): python: Import gz.math at startup to fix #1129  (#1130)

* python: Don't import more than is used in the tests - should cause a failure due to gz.math not always being imported.
* python: Import gz.math at startup to fix #1129
closed 2022-10-03 by ahcorde