gazebosim/sdformat #1261: ToElement methods do not export data from custom xml namespaces

scpeters, 2023-03-24:
## Environment
* OS Version: any
* Source or binary build? any

## Description
* Expected behavior: if you have an SDFormat file with data in custom XML namespaces (like [custom_elems_attrs.sdf](https://github.com/gazebosim/sdformat/blob/sdf13/test/integration/custom_elems_attrs.sdf)), if you load the file into an `sdf::Root` and then write it out again with `Root::ToElement()`, the resulting Element-tree should also have the custom data fields.
* Actual behavior: it does not include the custom data fields.

## Steps to reproduce

1. TODO: write code to reproduce this 
2.
3.

## Output

azeey, 2023-03-24:
I think `Root::Element()` would work though and maybe `ToElement` should check if the DOM object was loaded from XML (i.e, has a valid `Element()`) and return that.

scpeters, 2023-03-24:
> I think `Root::Element()` would work though and maybe `ToElement` should check if the DOM object was loaded from XML (i.e, has a valid `Element()`) and return that.

but if you make changes via the DOM API, will those be reflected in the `Element()` output?

azeey, 2023-03-24:
> but if you make changes via the DOM API, will those be reflected in the Element() output?

They would not. I guess if we want that we'd have to store the custom elements and attributes somehow and incorporate them into the output of `ToElement`. It wouldn't be trivial though since each element of the DOM object might have a custom attribute or custom element.

scpeters, 2023-03-27:
> They would not. I guess if we want that we'd have to store the custom elements and attributes somehow and incorporate them into the output of `ToElement`. It wouldn't be trivial though since each [...]

One approach would be to add `CustomData` class to the SDFormat c++ classes that stores a `Param_V` of custom attributes and an `ElementPtr_V` of custom child elements and populates it during parsing. Drawbacks:

* This would require touching every single c++ class.
* The data hierarchy is more complicated for classes that have child elements without their own c++ class, such as `//joint/axis/dynamics/*` and `//joint/axis/limit` that are stored within `sdf::JointAxis`

scpeters, 2024-05-02:
Based on [this portion of Plugin_TEST.cc](https://github.com/gazebosim/sdformat/blob/sdformat14_14.2.0/src/Plugin_TEST.cc#L242-L278), I believe that data from custom XML namespaces in the `Plugin::Contents` (child elements of `<plugin/>`) will be correctly exported by `Plugin::ToElement`. Exporting custom attributes of `<plugin/>` is not currently supported. In order to export custom attributes from a `<plugin/>` element for use in `gz-sim`, in addition to modifying the `sdf::Plugin` constructor in `sdf/Plugin.hh`, the corresponding [gz/msgs/plugin.proto](https://github.com/gazebosim/gz-msgs/blob/gz-msgs10/proto/gz/msgs/plugin.proto) message file would need to be updated as well.