gazebosim/sdformat #1276: :farmer: Sdf13 Windows `ModuleNotFoundError: No module named 'sdformat13'`

Crola1702, 2023-04-25:
## Environment
* OS Version: Windows
* Source Garden build

## Description
* Expected behavior: Test should pass
* Actual behavior: 44 tests fail

## Steps to reproduce
1. Run a build in https://build.osrfoundation.org/job/sdformat-sdf-13-win/
2. See tests fail.

## Output

Reference build: https://build.osrfoundation.org/job/sdformat-sdf-13-win/57/

Log output:
```
ImportError while importing test module 'C:\J\workspace\sdformat-sdf-13-win\ws\sdformat\python\test\pyAirPressure_TEST.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
C:\Program Files\Python311\Lib\importlib\__init__.py:126: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
..\..\..\sdformat\python\test\pyAirPressure_TEST.py:15: in <module>
    from gz_test_deps.sdformat import AirPressure, Noise
..\..\..\sdformat\python\test\gz_test_deps\sdformat.py:1: in <module>
    from sdformat13 import *
E   ModuleNotFoundError: No module named 'sdformat13'
```

It seems the python bindings aren't working as expected for the cloud windows agent. 

@azeey suggested disabling the bindings for this

j-rivero, 2023-04-25:
I was looking into the problem. There seems to be a two different types of problems:

1. More obvious: some sdformat python test files can not find gz-math python binding. This is not a problem on sdformat (or not only) but the root cause of it is that we are not generating python bindings for gz-math on Windows https://github.com/gazebosim/gz-math/issues/532

2. The `ModuleNotFoundError: No module named 'sdformat13'` error seems not so obvious to me. The python module is being installed in:
```
-- Installing: C:/J/workspace/sdformat-sdf-13-win/ws/install/sdformat13/lib/python/sdformat13.cp311-win_amd64.pyd
```
and the test PYTHONPATH seems to point to the same place:
```
233: Test command: "C:\Program Files\Python311\python.exe" "-m" "pytest" "C:/J/workspace/sdformat-sdf-13-win/ws/sdformat/python/test/pyAirPressure_TEST.py" "--junitxml" "C:/J/workspace/sdformat-sdf-13-win/ws/build/sdformat13/test_results/pyAirPressure_TEST.xml"
233: Environment variables: 
233:  PYTHONPATH=C:/J/workspace/sdformat-sdf-13-win/ws/install/sdformat13/lib/python/:C:/J/workspace/sdformat-sdf-13-win/ws/build/sdformat13/lib:
233:  LD_LIBRARY_PATH=C:/J/workspace/sdformat-sdf-13-win/ws/install/sdformat13
```
Might be problems with the separator character in PYTHONPATH for Windows or other kind of issues that I can not determine right now. Be careful if testing it manually due to problems with different python versions from system installation (3.11 and vcpkg 3.10).

azeey, 2023-05-02:
https://github.com/gazebosim/sdformat/pull/1278 disables pybind11 tests on windows until we find a solution for this issue.

JohnkaGL, 2023-07-21:
I got a similar error but on my Ubuntu jammy system, it seems that my OS can't recognise 'sdformat13' as a library file for python, i've tried to reinstall on binary and source distributions but nothing seems to work. this is the Error message i get from shell:

'Caught exception when trying to load file of format [py]: /usr/lib/python3/dist-packages/sdformat13.cpython-310-x86_64-linux-gnu.so: undefined symbol: _ZN3sdf3v139Projector10SetTextureERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE'

as I get to understand, its not creating the proper python file to make the wrapping... hope you can help!

aamodbk, 2023-10-28:
Any solution to fix this issue I'm having similar problems with running sdf2mjcf.

azeey, 2023-11-30:
@JohnkaGL and @aamodbk Is this still a problem? Have you tried updating the `libsdformat13` and `python3-sdformat13` packages. I tried locally on a fresh Jammy container and it seems to work fine.