gazebosim/sdformat #1285: XML Element[fullscreen], child of element[gui], not defined in SDF.

jrutgeer, 2023-05-22:
## Environment
* Ubuntu 22.04
* Source build, Gazebo Garden

SDFormat spec defines an element `<fullscreen>` in `<gui>`. 
However specifying this in an SDF (see example below) yields an error message: 
```
[/sdf/world[@name="fullscreen_test"]/gui/fullscreen:/home/johan/model.sdf:L6]: XML Element[fullscreen], child of element[gui], not defined in SDF. Copying[fullscreen] as children of [gui].
```

It seems that [not all GUI functionality is implemented](https://github.com/gazebosim/sdformat/blob/3fdc0a1fe7026fbc1f004de018daf83839a2914e/src/Gui.cc#L67), yet the strange thing is that on first sight [the fullscreen element _is_](https://github.com/gazebosim/sdformat/blob/3fdc0a1fe7026fbc1f004de018daf83839a2914e/src/Gui.cc#L60-L61).
So I assume it is rather an issue with the [parser](https://github.com/gazebosim/sdformat/blob/sdf13/src/parser.cc)?

On a side note: the `<origin_visual>` element doesn't seem to do anything either...

</br>

Corresponding SDF file:
```
<?xml version="1.0" ?>
<sdf version="1.9">
  <world name="fullscreen_test">

    <gui>
      <fullscreen>true</fullscreen>
    </gui>

    <scene>
      <origin_visual>true</origin_visual>
    </scene>

    <!-- Included model -->
    <include>
      <uri>https://fuel.gazebosim.org/1.0/OpenRobotics/models/Ground Plane</uri>
    </include>

 </world>
</sdf>
```

azeey, 2023-05-22:
`fullscreen` is defined to be an [attribute](https://github.com/gazebosim/sdformat/blob/e4fce4bd342feea3408fc08be261eaad938569a1/sdf/1.9/gui.sdf#L3) of `gui` instead of an element, so that explains why you're getting the error, but [website](http://sdformat.org/spec?ver=1.9&elem=world#gui_fullscreen) displays it as an element, which seems like a bug.

[image]

In fact, all attributes of description files that are included seem to be displayed as elements 😞

jrutgeer, 2023-05-23:
I changed it to `<gui fullscreen="true"> </gui>`, but it still opens in a non-maximized window.
A quick grep did not reveal any 'maximize window' calls, so I think this is not implemented after all.
Bummer. I will have to click on the icon. ;-)