gazebosim/sdformat #1300: fdir in URDF

muttistefano, 2023-07-14:
Hi all, 
i have been using gz-sim create to transport URDF files into gazebo and almost everything works nicely.
I have noticed that when I use mecanum omnidirectional robots, and I have to define a friction direction using the fdir keyowrd, it doesn't arrive to gazebo, and it gets lost somewhere in the middle.
Specifically, I have this in my URDF:
```
            <friction>
              <ode>
                <mu>1.0</mu>
                <mu2>0.0</mu2>
                <fdir1 gz:expressed_in="sweepee_1/base_footprint">1 -1 0</fdir1>
              </ode>
            </friction>
```
I noticed that the "gz:expressed_in="sweepee_1/base_footprint"" part is not received in the gazebo model, and hence it doesn't move correctly.
I checked the gz-sim package and it seems like the problem is elsewhere.
Is it because of the gz-transport package, is there something I can do to include this ? 
Thanks

azeey, 2023-07-14:
I was not able to reproduce this. What version of Gazebo are you using?

Here's the URDF I tested it on
```xml
<?xml version='1.0' encoding='UTF-8'?>
<robot name='test_robot'>
  <link name='base_link'>
    <inertial>
      <mass value='0.12' />
      <inertia ixx='0.01' ixy='0' ixz='0' iyy='0.01' iyz='0' izz='0.01' />
    </inertial>
    <collision>
      <geometry>
        <sphere radius="2"/>
      </geometry>
    </collision>
  </link>
  <gazebo reference='base_link'>
    <collision>
      <surface>
        <friction>
          <ode>
            <mu>1.0</mu>
            <mu2>0.0</mu2>
            <fdir1 gz:expressed_in="sweepee_1/base_footprint">1 -1 0</fdir1>
          </ode>
        </friction>
      </surface>
    </collision>
  </gazebo>
</robot>

```

Running `gz sdf -p path/to/test_robot.urdf` prints the following SDF that contains `gz:expressed_in="sweepee_1/base_footprint`:

```xml
<sdf version='1.10'>
  <model name='test_robot'>
    <link name='base_link'>
      <inertial>
        <pose>0 0 0 0 0 0</pose>
        <mass>0.12</mass>
        <inertia>
          <ixx>0.01</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>0.01</iyy>
          <iyz>0</iyz>
          <izz>0.01</izz>
        </inertia>
      </inertial>
      <collision name='base_link_collision'>
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <sphere>
            <radius>2</radius>
          </sphere>
        </geometry>
        <surface>
          <friction>
            <ode>
              <mu>1</mu>
              <mu2>0</mu2>
              <fdir1 gz:expressed_in='sweepee_1/base_footprint'>1 -1 0</fdir1>
            </ode>
          </friction>
          <contact>
            <ode/>
          </contact>
        </surface>
      </collision>
    </link>
  </model>
</sdf>
```

Can you try that on your system?

muttistefano, 2024-01-19:
I also cannot reproduce with newer gz version.

closed 2024-01-19 by muttistefano