gazebosim/sdformat #1321: gz sdf still not on Windows?

peci1, 2023-09-03:
https://github.com/gazebosim/sdformat/blob/881cbb7f4bab89d052095a43a11cf30059a174f4/src/CMakeLists.txt#L129-L131

Any reason why this is still there? What is the blocker?

peci1, 2023-09-03:
Bad thing is `gz sdf` command is actually registered to gz-tools, so it is presented in help messages etc., but the ruby script is not installed, so any attempt at running it fails with a cryptic error.

traversaro, 2023-09-03:
See https://github.com/gazebosim/gz-tools/issues/71 for a related discussion. In a nutshell we need to propagate the fix https://github.com/gazebosim/gz-plugin/pull/63 here, just removing the if if I am not wrong will break builds on Visual Studio multiple-config generators.

peci1, 2023-09-04:
> if if I am not wrong will break builds on Visual Studio multiple-config generators.

You're right ;)

commit e9a2c01d (2024-05-10): Fix macOS workflow and backport windows fix (#1409)

Partially backports #1374 to fix #1321
* Add package.xml, fix `gz sdf` tests on Windows (#1374)
  The `gz sdf` tests are fixed by
    * fixing dll path issue with Ruby on windows
    * Setting home path
* Fix macOS workflow
closed 2024-05-31 by iche033