gazebosim/sdformat #1384: Consider adding `assert_convex` to mesh specification

scpeters, 2024-03-21:
## Desired behavior

The `//mesh/@optimization` attribute is being added in https://github.com/gazebosim/sdformat/pull/1382 with recognition for 3 values of the attribute:

* `""` (empty string): no mesh optimization is done
* `convex_hull`: use the convex hull of the specified mesh
* `convex_decomposition`: decompose the mesh into convex sub-meshes (using parameters from `//mesh/convex_decomposition`)

It could also be useful for a user to specify `assert_convex` when they believe a mesh is convex and that an error should be raised if it is found to not be convex. Verifying this would require code from gz-common (or similar) to get the mesh data for analysis, so it couldn't be added directly to `sdformat`, but it could be useful for some users.

There is a use case for wanting to specify both `assert_convex` and `convex_hull` (a user may want to use the `convex_hull` optimization with a mesh known to be convex since it may consolidate triangle faces into larger polygon faces), so I would suggest a new attribute `//mesh/@assert_convex`.

## Alternatives considered

Instead of a new attribute, the `assert_convex` behavior could be specified in `//mesh/@optimization`.

Also, I've mentioned this as a potential use case, but I'm not sure how much user demand there actually is. The `convex_hull` option may be adequate, combined with a CLI tool in another repository for evaluating how close to convex a given mesh is.

## Implementation suggestion

## Additional context

Thanks to @SeanCurtis-TRI for suggesting the `assert_convex` name and discussing this potential use case.

shameekganguly, 2025-08-06:
@scpeters I'd like to volunteer to add this feature in sdformat and follow-up with consuming it in gz-physics and gz-sim.

The main use case is to use fine-tuned convex decomposition generated from an external tool as a set of collision geometry meshes for a link in sdformat. Currently, we can annotate each convex mesh with the `convex_hull` optimization attribute, but this still leads to a non-trivial amount of sim start-up time spent on "convex decomposition" of each mesh (~2-3s per mesh, which can accumulate to several minutes for loading an object with hundreds of convex parts).

My main concern with adding a new `//mesh/@assert_convex` attribute is that checking if an input mesh is convex has about the same average time complexity as computing the convex hull (assuming we use [quickhull](https://en.wikipedia.org/wiki/Quickhull) to compute the convex hull, then check if each vertex on the original mesh lies on the face of the hull). So users might as well just set the `convex_hull` optimization.

I'd also like to propose instead to add `assert_convex` as a new value within the `//mesh/@optimization` attribute (your suggested alternative) which would indicate to consumers that the mesh is already convex, allowing specialized handling such as for collision checking in physics. Additionally, we don't have to deal with the combinatorial complexity of multiple related attributes under `//mesh/`. The risk here of course is that users can accidentally pass in a non-convex mesh with `//mesh/@optimization="assert_convex"`, leading to hard to debug issues such as unexpected collisions (unless simulators also build additional features to help debug such issues).

Another alternative would be add a new `//mesh/@assert_convex` attribute, but make it string-enum-valued with the following allowed values:
* `nocheck`: use mesh as convex without checking.
* `warn`: check if mesh is convex and warn if not. If it is not convex, the computed convex hull will be used.
* `error`: check if mesh is convex and error out if it is not.

There is another consideration about dealing with meshes with multiple sub-meshes. Again, considering the use-case of using an externally generated fine-tuned convex decomposition collision mesh, it would be optimal to export the decomposition as a single mesh with convex sub-meshes (e.g. single .glb file, to reduce I/O overhead). But the current behavior of the `convex_hull` optimization is to apply it to the whole mesh rather than to individual sub-meshes. 
* If we go with adding `assert_convex` to the existing `optimization` attribute, I'd propose we add a separate `assert_convex_parts` value to treat each sub-mesh as convex. 
* If we add `assert_convex` as a new attribute, I'd propose we add a separate `apply_optimization_to` attribute to specify if the optimization should be applied to the entire mesh or to individual sub-meshes.

CC @iche033 for your thoughts

iche033, 2025-08-06:
> There is a use case for wanting to specify both assert_convex and convex_hull (a user may want to use the convex_hull optimization with a mesh known to be convex since it may consolidate triangle faces into larger polygon faces)

Trying to understand if having both `assert_convex` and `convex_hull` is necessary. One possible use case I can think of is that the user would only want to create a convex hull of the mesh (`//mesh/@optimization='convex_hull'`) only if the original input mesh is already convex (using `//mesh/@assert_convex`).

If the above use case is important, I would lean for the alternative approach suggested by @shameekganguly that uses string enums for the `//mesh/@assert_convex` attribute. It's also more explicit and gives users a finer control on the behavior of mesh optimization