gazebosim/sdformat #1385: Consider declare meter-kilogram-second officially as part of the specification.

qingyouzhao, 2024-03-25:
## Context

In various documentations in the SDF spec, units kilograms, meters and seconds are used for different physical properties.
For example:
- [kg and m in inertial](https://github.com/gazebosim/sdformat/blob/sdf14/sdf/1.11/inertial.sdf#L116-L232)
- [Meters in Specifying pose tutorial](https://github.com/gazebosim/sdformat/blob/sdf14/sdf/1.11/inertial.sdf#L116-L232)
However there is not a centralized place that states these are the official units of the sdformat. 

## Problem

This created some ambiguity around inferring units:
1. In places where units are not clearly stated, for example in [geometry](https://github.com/gazebosim/sdformat/blob/sdf14/sdf/1.11/geometry.sdf), should one assume all geometries are defined in meters?
2. In a included mesh in [mesh_shape](https://github.com/gazebosim/sdformat/blob/sdf14/sdf/1.11/mesh_shape.sdf), should one assume mesh vertex locations are in meters even if the included mesh file might have another unit convention. For example, FBX files have [implied centimeters](https://forums.autodesk.com/t5/fbx-forum/fbx-units-and-scaling/td-p/9761576) although it is stated as an unit-less format.

## Proposal

Let's discuss options. There are a few options I can think. 
1. Define units in a centralized place and only document exceptions in individual schemas. This is the simplest way to disambiguate if everyone has been working with this assumption.
2. Define units explicitly as part of the schema similar to [angle vs radians in pose](https://github.com/gazebosim/sdformat/blob/sdf14/sdf/1.11/pose.sdf#L37). This would be more tedious to implement but makes units more explicit

scpeters, 2024-04-01:
> Let's discuss options. There are a few options I can think of.
> 
> 1. Define units in a centralized place and only document exceptions in individual schemas. This is the simplest way to disambiguate if everyone has been working with this assumption.
> 2. Define units explicitly as part of the schema similar to [angle vs radians in pose](https://github.com/gazebosim/sdformat/blob/sdf14/sdf/1.11/pose.sdf#L37). This would be more tedious to implement but makes units more explicit

I prefer option 1, and the we refer to [ROS REP 103](https://www.ros.org/reps/rep-0103.html), which defines standard units for ROS.

azeey, 2024-05-06:
I'd prefer option 1 as well. I think for mesh files, the mesh loader should be implemented such that if the file format has a convention of using a different unit (e.g. centimeters), it would convert the values to meters during import.