gazebosim/sdformat #1411: Allow Xacro files to be included in world

Jack-ReframeSystems, 2024-05-08:
## Desired behavior

I have a couple of robots expressed as Xacro files. I would like to be able to include these in my Simulation world by using:
```
<world>
  ...xml
  <include>
     <uri>
        model://my_robot
    </uri>
    <pose> ... </pose>
  <include>
  ...
</world>
```
I have a `model.config` for my_robot that looks like:
```
<?xml version="1.0"?>
<model>
  <name>my_robot</name>
  <version>1.0</version>
  <sdf version='1.5'>my_robot.xacro</sdf>

  <author>
    <name>Jack</name>
    <email> ... </email>
  </author>

  <description>
    ...
  </description>
</model>
```

I would expect Gazebo (and therefore SDFormat) to be able to handle Xacro files. I would expect the change to need to be supported here, just like URDF to SDF conversion is done before loading an SDF.

https://github.com/gazebosim/sdformat/blob/6f1c36502f1085836ab9876e26afac3238f26820/src/parser.cc#L859-L860

## Alternatives considered

The current solution of running Xacro before inserting the model into Gazebo. The issue with this approach is that I have a lot of models and I don't want to have to write custom code to go through, convert to URDF, and then dynamically load into Gazebo. I think this feature would improve the ergonomics of GzSim and SDFormat.

## Implementation suggestion

Adding another if statement at the line linked above to prerun Xacro (if on the system) if the file ends in `.xacro`. We also might want to start structuring the decision of which file to convert to as a more patterned approach than an if statement. Open to suggestions on how you would structure this.

### Additional Context

I would like to add this feature myself, but I want to get the projects thoughts on whether this is the right place to add this feature / if it would be accepted.

azeey, 2024-05-24:
Supporting xacro files would be great and I like the idea of prerunning `xacro`  on `.xacro` files. I would think the change to go https://github.com/gazebosim/sdformat/blob/6f1c36502f1085836ab9876e26afac3238f26820/src/parser.cc#L843 right before `xmlDoc.LoadFile`, so that the output of xacro is fed into `xmlDoc.LoadString` instead. We'll need to handle errors well, so that it's clear whether the error is coming from `xacro` or the resulting URDF or SDF file.

> We also might want to start structuring the decision of which file to convert to as a more patterned approach than an if statement.
Are you saying we should make it more extensible to other preprocessors like xacro? Other than `.erb`, I'm not sure if there is any other commonly used  format, so I'm not sure if it's worth creating more complexity here. But it would be cool to do the same for `.erb` files and prerun `erb`.

azeey, 2024-07-02:
@Jack-ReframeSystems If you're still planning to work on this, I can assign this issue to you.