gazebosim/sdformat #1417: Add python bindings for auto-inertial APIs added in sdf14

scpeters, 2024-05-15:
## Desired behavior

There should by python bindings for the C++ APIs added to support the automatic calculation of inertial properties in `sdf14`. I believe the following pull requests include the APIs added for this feature:

* https://github.com/gazebosim/sdformat/pull/1299
* https://github.com/gazebosim/sdformat/pull/1304
* https://github.com/gazebosim/sdformat/pull/1335

## Alternatives considered

## Implementation suggestion

## Additional context

ahcorde, 2024-05-28:
Related with the first PR https://github.com/gazebosim/sdformat/pull/1424

pranay172, 2026-10-22:
Hi @scpeters, I'd like to work on this. Comparing `python/src/sdf` on `main` with the APIs from #1304 and #1335, these still look unbound:

- `CustomInertiaCalcProperties` (new class)
- `ParserConfig::RegisterCustomInertiaCalc` / `CustomInertiaCalc`
- `Mesh::CalculateInertial`
- `Link::Density` / `SetDensity` / `AutoInertiaParams` / `SetAutoInertiaParams`
- `Collision::DensityDefault` / `AutoInertiaParams` / `SetAutoInertiaParams`, and the `CalculateInertial` overload that takes a density and auto-inertia params

`ParserConfig::CalculateInertialFailurePolicy` and its enum also seem unbound. Should I include them here as well? I plan to target `main` and add Python tests for each new binding. Does that scope sound right?