gazebosim/sdformat #1485: No specification for v12, v13, or v14 + general documentation request

benbaarber, 2024-10-04:
I have been working with gazebo harmonic which ships with sdf v14. There seems to be helpful specification on sdformat.org for v11 and prior versions, but none at all for any of the newer versions. I am continuously running into issues with unsupported and undocumented elements. How am I or anyone else using gazebo supposed to know all the ins and outs of the newer SDF versions if there are no docs? Where do I find this information? Am I expected to read the source code? Not even being sarcastic, genuinely just wondering what me and other users are supposed to do here. 

I know open source moves fast, and can be unforgiving work, and I commend the open robotics team for creating such a powerful open source ecosystem for robotics. That said, I have worked with many open source tools, and gazebo and its supporting tools have the steepest learning curve and the most pitfalls and missing or incorrect documentation I have ever seen. I think it would be *massively* beneficial to your project, and especially to your users, to slow down a bit on the new features and lean in heavily to documenting existing ones. 

I'm a huge fan of the work you all have put into this, and I think I speak for a lot of people when I say we need docs. The most powerful tool in the world could still be useless if nobody knows how to use it.

benbaarber, 2024-10-07:
Also, I keep getting this error: `[Err] [SDFFeatures.cc:1184] Asked to construct a joint of sdf::JointType [2], but that is not supported yet. Creating a FIXED joint instead`, when trying to create a continuous joint in the sdf file. Have not been able to find anything about this either. Is it a lack of support in a recent version of one of the physics engines? I believe I am using either dart or ogre2.

Amronos, 2024-11-03:
@benbaarber please have a look at what's written at the top of [this](http://sdformat.org/spec) page under the "Versions" heading .
IMO the specification for these versions of sdformat is well documented.
[This](http://sdformat.org/tutorials?tut=roadmap) page can be improved with the latest info though.