gazebosim/sdformat #1518: Dynamic Link error: undefined symbol: _ZN3sdf3v144Root20ClearActorLightModelEv in sdformat14

janugent, 2024-12-14:
## Environment
* OS Version: Ubuntu 24.04
* Happens with Binary and source Install

## Description
* Expected behavior: Dynamic linking success 
* Actual behavior:  Fails with this error:  undefined symbol: _ZN3sdf3v144Root20ClearActorLightModelEv in sdformat14

## Steps to reproduce
0. Install required dependencies:  sudo apt -y install $(sort -u $(find . -iname 'packages-'`lsb_release -cs`'.apt' -o -iname 'packages.apt' | grep -v '/\.git/') | tr '\n' ' ')
1. If the library is installed,   python3 -c "import sdformat14"  should do it.
2. Also can be seen in the source distribution by doing "make test" after building

## Output
In python3 doing
>import sdformat14
`james@tiasln:~/seaqr/sdformat/build$ python3 -c "import sdformat14"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: /usr/lib/python3/dist-packages/sdformat14.cpython-312-x86_64-linux-gnu.so: undefined symbol: _ZN3sdf3v144Root20ClearActorLightModelEv
`

With the source distribution, doing 'make test' after compiling gives :
`Test project /home/james/seaqr/sdformat/build
    Start 101: UNIT_Root_TEST
1/1 Test #101: UNIT_Root_TEST ...................***Failed    0.00 sec
/home/james/seaqr/sdformat/build/bin/UNIT_Root_TEST: symbol lookup error: /home/james/seaqr/sdformat/build/bin/UNIT_Root_TEST: undefined symbol: _ZN3sdf3v144Root20ClearActorLightModelEv
`
All other test pass however.

traversaro, 2024-12-14:
Are you sure you do not have already a sdformat14 installed with apt?

janugent, 2024-12-16:
I got this with both the apt version and the source distribution. I uninstalled the apt version and installed the source version and had the same issue.   I also got it  with the unit test in the repo when compiling the source version, which shouldn't use the installed version. (and I didn't have the apt version then anyway)

traversaro, 2024-12-16:
Are you sure you do not have an old from-source version installed in `/usr/local` or similar? Can you report the output of `ldd /usr/lib/python3/dist-packages/sdformat14.cpython-312-x86_64-linux-gnu.so` ?

janugent, 2024-12-16:
There isn't any other version. (It's a new ubuntu install)

`james@tiasln:~/seaqr/sdformat$ whereis sdformat14
sdformat14: /usr/share/sdformat14
`

`james@tiasln:~/seaqr/sdformat$ ldd /usr/lib/python3/dist-packages/sdformat14.cpython-312-x86_64-linux-gnu.so
	linux-vdso.so.1 (0x00007fffe09f0000)
	libsdformat14.so.14 => /opt/ros/jazzy/opt/sdformat_vendor/lib/libsdformat14.so.14 (0x000072c01bc00000)
	libgz-math7.so.7 => /opt/ros/jazzy/opt/gz_math_vendor/lib/libgz-math7.so.7 (0x000072c01bfa6000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x000072c01b800000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x000072c01be92000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x000072c01bbd2000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x000072c01b400000)
	/lib64/ld-linux-x86-64.so.2 (0x000072c01c26f000)
	libgz-utils2.so.2 => /opt/ros/jazzy/opt/gz_utils_vendor/lib/libgz-utils2.so.2 (0x000072c01be86000)
	libtinyxml2.so.10 => /lib/x86_64-linux-gnu/libtinyxml2.so.10 (0x000072c01bbbb000)
	liburdfdom_model.so.4.0 => /opt/ros/jazzy/lib/x86_64-linux-gnu/liburdfdom_model.so.4.0 (0x000072c01bb9f000)
	libconsole_bridge.so.1.0 => /lib/x86_64-linux-gnu/libconsole_bridge.so.1.0 (0x000072c01bb99000)
`

azeey, 2024-12-16:
It looks like you have SDFormat installed from packages.osrfoundation.org, but also from packages.ros.org via `ros-jazzy-sdformat-vendor`. The two versions of SDFormat might be different at a given time point because when a new version of SDFormat is released, it goes into packages.osrfoundation.org immediately, however, there's a short period of time (a couple weeks or so) before it makes it into packages.ros.org. I think this explains the problem you're facing. My suggestion is you try  running `python3 -c "import sdformat14" ` without sourcing your ROS workspace.

janugent, 2024-12-23:
Thanks azeey!  That fixed it.  I wiped out ROS and Gazebo and reinstalled making sure there were no duplicates. Didn't actually fix the problem but installed the source version again and discovered there was  minor version difference  between the vendor package and here.   It seems like there was a bug with the version in the vendor  packages (14.5.0) that  was fixed here(14.6.0 was fine) but had not migrated over yet. Sorry about that, seems like the issue  was already fixed. I'm sorry about reporting it then.  :(

janugent, 2024-12-23:
Closing due to being already fixed.

closed 2024-12-23 by janugent