gazebosim/sdformat #1534: Gazebo material format

Tacha-S, 2025-01-31:
SDF allows the use of `script` within the `material` definition, and in `gz simulator`, the `material` tag inside the `gazebo` tag also supports `script`.  
However, when specifying it as follows, `gz simulator` requests a string upon startup.

http://sdformat.org/spec?ver=1.12&elem=material
https://github.com/gazebosim/sdformat/blob/2b583cbc3f40ac4f1af27613a49b475f2d0d99b7/src/parser_urdf.cc#L1317-L1321

```xml
<gazebo reference="link1">
      <material>
        <script>
          <uri>file://media/materials/scripts/gazebo.material</uri>
          <name>Gazebo/Black</name>
        </script>
      </material>
    </gazebo>
```

## Environment
* OS Version: Ubuntu 24.04
* Source or binary build: Binary 8.8.0

## Description
* Expected behavior: 
   NO warning.
* Actual behavior: 
  `Attribute value string not set` is shown.

## Steps to reproduce

1. Set material color to a link
```xml
<gazebo reference="link1">
      <material>
        <script>
          <uri>file://media/materials/scripts/gazebo.material</uri>
          <name>Gazebo/Black</name>
        </script>
      </material>
    </gazebo>
```
2. launch gz server

## Output

```bash
[gazebo-1] Warning [parser_urdf.cc:1220] Attribute value string not set
[gazebo-1] Warning [parser_urdf.cc:1220] Attribute value string not set
[gazebo-1] Warning [parser_urdf.cc:1220] Attribute value string not set
[gazebo-1] Warning [parser_urdf.cc:1220] Attribute value string not set
[gazebo-1] Warning [parser_urdf.cc:1220] Attribute value string not set
[gazebo-1] Warning [parser_urdf.cc:1220] Attribute value string not set
[gazebo-1] Warning [parser_urdf.cc:1220] Attribute value string not set
[gazebo-1] Warning [parser_urdf.cc:1220] Attribute value string not set
[gazebo-1] Warning [parser_urdf.cc:1220] Attribute value string not set
[gazebo-1] Warning [parser_urdf.cc:1220] Attribute value string not set
[gazebo-1] Warning [parser_urdf.cc:1220] Attribute value string not set
```

azeey, 2025-01-31:
Can you please provide a full `.sdf` or `.urdf` file to try?

Tacha-S, 2025-02-01:
example.urdf
```xml
<?xml version="1.0" encoding="UTF-8"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="bot">
  <link name="base_footprint"/>
  <joint name="base_joint" type="fixed">
    <parent link="base_footprint"/>
    <child link="base_link"/>
    <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
  </joint>

  <link name="base_link">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="1 1 1"/>
      </geometry>
      <material name="white">
        <color rgba="1 1 1 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="1 1 1"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="1"/>
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <inertia ixx="0.1" ixy="0.0" ixz="0.0" iyy="0.1" iyz="0.0" izz="0.1"/>
    </inertial>
  </link>
  <gazebo reference="base_link">
    <material>
      <script>
        <uri>file://media/materials/scripts/gazebo.material</uri>
        <name>Gazebo/Black</name>
      </script>
    </material>
  </gazebo>
</robot>
```

1. launch gz server
`gz sim empty.sdf`
2. spawn example urdf
`gz service -s /world/empty/create --reqtype gz.msgs.EntityFactory --reptype gz.msgs.Boolean --timeout 1000 --req 'sdf_filename: "/path/to/example.urdf", name: "bot"'`