gazebosim/sdformat #1578: Floats using D parse without warning but break later

EmmanuelMess, 2025-07-24:
FORTRAN allows use of the character "D"/"d" instead of "E"/"e" in floats, it looks like the parser doesn't warn the user when this is done in sdf, but it doesn't seem to parse it correctly. This notation is used extensively in GPS files, and the incorrect input should be warned or should fail to parse.

scpeters, 2025-09-09:
I think we can pass a `std::size_t*` pointer as the second argument of `std::stod` to determine how many characters were used as part of the conversion and report a warning / error if some characters were not included

can you share which datatypes are most susceptible to this issue, such as Scalars / Vector3 / Pose3?

EmmanuelMess, 2025-09-09:
I am using a custom parsing routine in my plugin,that parses a custom tag:
```
            <gz:klobuchar>
              <gz:alphas>
                <gz:coeff0>0.6519e-8</gz:coeff0>
                <gz:coeff1>0.1490e-7</gz:coeff1>
                <gz:coeff2>-0.5960e-7</gz:coeff2>
                <gz:coeff3>-0.1192e-6</gz:coeff3>
              </gz:alphas>
              <gz:betas>
                <gz:coeff0>0.7782e5</gz:coeff0>
                <gz:coeff1>0.3277e5</gz:coeff1>
                <gz:coeff2>-0.6554e5</gz:coeff2>
                <gz:coeff3>-0.1966e6</gz:coeff3>
              </gz:betas>
            </gz:klobuchar>
```
Read like this:
```
  const auto alpha0 = alphasElement->FindElement(SDF_COEFF_0);
  const auto alpha1 = alphasElement->FindElement(SDF_COEFF_1);
  const auto alpha2 = alphasElement->FindElement(SDF_COEFF_2);
  const auto alpha3 = alphasElement->FindElement(SDF_COEFF_3);
  const auto beta0 = betasElement->FindElement(SDF_COEFF_0);
  const auto beta1 = betasElement->FindElement(SDF_COEFF_1);
  const auto beta2 = betasElement->FindElement(SDF_COEFF_2);
  const auto beta3 = betasElement->FindElement(SDF_COEFF_3);

  this->klobucharAlphas  = { alpha0->Get<double>(), alpha1->Get<double>(), alpha2->Get<double>(), alpha3->Get<double>()};
  this->klobucharBetas = { beta0->Get<double>(), beta1->Get<double>(), beta2->Get<double>(), beta3->Get<double>()};
```

I don't this file will be useful to you, because of the read code needed.

scpeters, 2025-09-09:
so I assume some of the `gz:coeff*` elements are the ones where the `d` syntax may be used? We will be sure to add tests to cover this case

EmmanuelMess, 2025-09-09:
> so I assume some of the `gz:coeff*` elements are the ones where the `d` syntax may be used? We will be sure to add tests to cover this case

I would recommend you simply add an error for the double parsing, and not try to parse d, as it might be more confusing to have only part of the FORTRAN parsing of floating point formats.